robocomp/robocomp #5: Document

Lonelygeek, 2015-03-07:
In the  documentation it is given that  
You might also want these nice developer tools:

sudo apt-get kdevelop vim aptitude nano 

but

sudo apt-get install kdevelop vim aptitude nano 

would be more suitable

pbustos, 2015-03-07:
Thanks!

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

Lonelygeek, 2015-03-07:
While i am trying cmak . it shows following error

seems like it is due to  the bug in cmake should i fix it /If not anything wrong with my steps ?

chandy@chandy:~$ cd robocomp
chandy@chandy:~/robocomp$ sudo cmake .
[sudo] password for chandy: 
-- Not found OpenMP
-- PYTHON BINDINGS:        NO
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- COMPILING WITHOUT FCL SUPPORT  = 0
CMake Error at tools/rcinnerModelSimulator/src/CMakeLists.txt:44 (INCLUDE):
  include could not find load file:

    /cmake/robocomp.cmake

CMake Error at tools/rcinnerModelSimulator/src/CMakeLists.txt:45 (ROBOCOMP_INITIALIZE):
  Unknown CMake command "ROBOCOMP_INITIALIZE".

-- Configuring incomplete, errors occurred!
See also "/home/chandy/robocomp/CMakeFiles/CMakeOutput.log".
See also "/home/chandy/robocomp/CMakeFiles/CMakeError.log".
chandy@chandy:~/robocomp$

pbustos, 2015-03-07:
Hello,

you need the ROBOCOMP variable defined for your terminal (bash)

Pablo

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

Lonelygeek, 2015-03-07:
Yes i have done that 

export ROBOCOMP=/home/chandy/robocomp
export PATH=$PATH:/opt/robocomp/bin

but shows same error

pbustos, 2015-03-07:
Why

    sudo cmake .

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

Lonelygeek, 2015-03-07:
When try cmake . it shows permission issues and shows following error 

chandy@chandy:~/robocomp$ cmake .
-- Not found OpenMP
-- PYTHON BINDINGS:        NO
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
  file Internal CMake error when trying to open file:
  /home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_adapterwidget.cxx_parameters
  for writing.
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
  libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)

CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
  file Internal CMake error when trying to open file:
  /home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_findnamednode.cxx_parameters
  for writing.
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
  libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)

CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
  file Internal CMake error when trying to open file:
  /home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_getworldcoorofnode.cxx_parameters
  for writing.
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
  libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)

CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
  file Internal CMake error when trying to open file:
  /home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_osgview.cxx_parameters
  for writing.
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
  libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)

CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
  file Internal CMake error when trying to open file:
  /home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_viewerqt.cxx_parameters
  for writing.
Call Stack (most recent call first):
  /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
  libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)

-- COMPILING WITHOUT FCL SUPPORT  = 0
-- RoboComp root is now set to /home/chandy/robocomp/
-- $ICEROOT=""
-- 
-- 
-- 
-- $SLICE_PATH=";/opt/robocomp/ThirdPartyinterfaces;/home/chandy/robocomp//interfaces/"
-- Adding /opt/robocomp/ThirdPartyinterfaces to the Slice directory set.
-- Adding /home/chandy/robocomp//interfaces/ to the Slice directory set.
-- Adding rule to generate Camera.cpp and Camera.h from /home/chandy/robocomp//interfaces//Camera.ice  (slice2cpp)
-- Adding rule to generate CommonBehavior.cpp and CommonBehavior.h from /home/chandy/robocomp//interfaces//CommonBehavior.ice  (slice2cpp)
-- Adding rule to generate CommonHead.cpp and CommonHead.h from /home/chandy/robocomp//interfaces//CommonHead.ice  (slice2cpp)
-- Adding rule to generate DifferentialRobot.cpp and DifferentialRobot.h from /home/chandy/robocomp//interfaces//DifferentialRobot.ice  (slice2cpp)
-- Adding rule to generate OmniRobot.cpp and OmniRobot.h from /home/chandy/robocomp//interfaces//OmniRobot.ice  (slice2cpp)
-- Adding rule to generate IMU.cpp and IMU.h from /home/chandy/robocomp//interfaces//IMU.ice  (slice2cpp)
-- Adding rule to generate InnerModelManager.cpp and InnerModelManager.h from /home/chandy/robocomp//interfaces//InnerModelManager.ice  (slice2cpp)
-- Adding rule to generate JointMotor.cpp and JointMotor.h from /home/chandy/robocomp//interfaces//JointMotor.ice  (slice2cpp)
-- Adding rule to generate Laser.cpp and Laser.h from /home/chandy/robocomp//interfaces//Laser.ice  (slice2cpp)
-- Adding rule to generate RGBD.cpp and RGBD.h from /home/chandy/robocomp//interfaces//RGBD.ice  (slice2cpp)
-- Adding rule to generate TouchSensor.cpp and TouchSensor.h from /home/chandy/robocomp//interfaces//TouchSensor.ice  (slice2cpp)
-- RoboComp libraries
-- Library qmat not found in /opt/robocomp/lib
CMake Error at cmake/robocomp.cmake:62 (MESSAGE):
  Library qmat not found
Call Stack (most recent call first):
  tools/rcinnerModelSimulator/src/CMakeLists.txt:47 (ROBOCOMP_LIBRARY)

-- Configuring incomplete, errors occurred!
See also "/home/chandy/robocomp/CMakeFiles/CMakeOutput.log".
See also "/home/chandy/robocomp/CMakeFiles/CMakeError.log".
CMake Error: Unable to open check cache file for write. /home/chandy/robocomp/CMakeFiles/cmake.check_cache

pbustos, 2015-03-07:
All files below Robocomp shoul belong to you. Looks  like you used sudo at
some point and created root owned ones. Clean and clone again.

> When try cmake . it shows permission issues and shows following error
>
> chandy@chandy:~/robocomp$ cmake .
>
> -- Not found OpenMP
> -- PYTHON BINDINGS: NO
> -- Could NOT find Doxygen (missing: [...]
/home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_adapterwidget.cxx_parameters
> for writing.
> Call Stack (most recent call first):
> /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
> libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)
>
> CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
> file Internal CMake error when trying to open file:
/home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_findnamednode.cxx_parameters
> for writing.
> Call Stack (most recent call first):
> /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
> libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)
>
> CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
> file Internal CMake error when trying to open file:
/home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_getworldcoorofnode.cxx_parameters
> for writing.
> Call Stack (most recent call first):
> /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
> libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)
>
> CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
> file Internal CMake error when trying to open file:
/home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_osgview.cxx_parameters
> for writing.
> Call Stack (most recent call first):
> /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
> libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)
>
> CMake Error at /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:131 (file):
> file Internal CMake error when trying to open file:
/home/chandy/robocomp/libs/osgviewer/include/osgviewer/moc_viewerqt.cxx_parameters
> for writing.
> Call Stack (most recent call first):
> /usr/share/cmake-2.8/Modules/Qt4Macros.cmake:170 (QT4_CREATE_MOC_COMMAND)
> libs/osgviewer/CMakeLists.txt:28 (QT4_WRAP_CPP)
>
> -- COMPILING WITHOUT FCL SUPPORT = 0
> -- RoboComp root is now set to /home/chandy/robocomp/
>
> -- $ICEROOT=""
$SLICE_PATH=";/opt/robocomp/ThirdPartyinterfaces;/home/chandy/robocomp//interfaces/"
> -- Adding /opt/robocomp/ThirdPartyinterfaces to the Slice directory set.
> -- Adding /home/chandy/robocomp//interfaces/ to the Slice directory set.
> -- Adding rule to generate Camera.cpp and Camera.h from
/home/chandy/robocomp//interfaces//Camera.ice (slice2cpp)
> -- Adding rule to generate CommonBehavior.cpp and CommonBehavior.h from
/home/chandy/robocomp//interfaces//CommonBehavior.ice (slice2cpp)
> -- Adding rule to generate CommonHead.cpp and CommonHead.h from
/home/chandy/robocomp//interfaces//CommonHead.ice (slice2cpp)
> -- Adding rule to generate DifferentialRobot.cpp and DifferentialRobot.h
from /home/chandy/robocomp//interfaces//DifferentialRobot.ice (slice2cpp)
> -- Adding rule to generate OmniRobot.cpp and OmniRobot.h from
/home/chandy/robocomp//interfaces//OmniRobot.ice (slice2cpp)
> -- Adding rule to generate IMU.cpp and IMU.h from
/home/chandy/robocomp//interfaces//IMU.ice (slice2cpp)
> -- Adding rule to generate InnerModelManager.cpp and InnerModelManager.h
from /home/chandy/robocomp//interfaces//InnerModelManager.ice (slice2cpp)
> -- Adding rule to generate JointMotor.cpp and JointMotor.h from
/home/chandy/robocomp//interfaces//JointMotor.ice (slice2cpp)
> -- Adding rule to generate Laser.cpp and Laser.h from
/home/chandy/robocomp//interfaces//Laser.ice (slice2cpp)
> -- Adding rule to generate RGBD.cpp and RGBD.h from
/home/chandy/robocomp//interfaces//RGBD.ice (slice2cpp)
> -- Adding rule to generate TouchSensor.cpp and TouchSensor.h from
/home/chandy/robocomp//interfaces//TouchSensor.ice (slice2cpp)
> -- RoboComp libraries
> -- Library qmat not found in /opt/robocomp/lib
> CMake Error at cmake/robocomp.cmake:62 (MESSAGE):
> Library qmat not found
> Call Stack (most recent call first): [...]
/home/chandy/robocomp/CMakeFiles/cmake.check_cache

Lonelygeek, 2015-03-13:
Now it's working fine

closed 2015-03-21 by Lonelygeek