robocomp/robocomp #7: CMake failed: Library qmat not found in /opt/robocomp/lib

hustcalm, 2015-03-09:
Follow the instructions [here - Installation in Ubuntu](https://github.com/robocomp/robocomp), my computer is running Ubuntu 14.04.

    export ROBOCOMP=<installation-directory>/robocomp
    export PATH=$PATH:/opt/robocomp/bin

I clone the source code to `/home/lihang/robocomp`, so I just do:

    export ROBOCOMP=home/lihang/robocomp

Then when running `cmake .`, the output is as following lines:

```bash
-- Not found OpenMP
-- PYTHON BINDINGS:        NO
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake-qt4 (found version "4.8.6")
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6")
-- COMPILING WITHOUT FCL SUPPORT  = 0
-- RoboComp root is now set to /home/lihang/robocomp/
-- $ICEROOT=""
--
--
--
-- $SLICE_PATH=";/opt/robocomp/ThirdPartyinterfaces;/home/lihang/robocomp//interfaces/"
-- Adding /opt/robocomp/ThirdPartyinterfaces to the Slice directory set.
-- Adding /home/lihang/robocomp//interfaces/ to the Slice directory set.
-- Adding rule to generate Camera.cpp and Camera.h from /home/lihang/robocomp//interfaces//Camera.ice  (slice2cpp)
-- Adding rule to generate CommonBehavior.cpp and CommonBehavior.h from /home/lihang/robocomp//interfaces//CommonBehavior.ice  (slice2cpp)
-- Adding rule to generate CommonHead.cpp and CommonHead.h from /home/lihang/robocomp//interfaces//CommonHead.ice  (slice2cpp)
-- Adding rule to generate DifferentialRobot.cpp and DifferentialRobot.h from /home/lihang/robocomp//interfaces//DifferentialRobot.ice  (slice2cpp)
-- Adding rule to generate OmniRobot.cpp and OmniRobot.h from /home/lihang/robocomp//interfaces//OmniRobot.ice  (slice2cpp)
-- Adding rule to generate IMU.cpp and IMU.h from /home/lihang/robocomp//interfaces//IMU.ice  (slice2cpp)
-- Adding rule to generate InnerModelManager.cpp and InnerModelManager.h from /home/lihang/robocomp//interfaces//InnerModelManager.ice  (slice2cpp)
-- Adding rule to generate JointMotor.cpp and JointMotor.h from /home/lihang/robocomp//interfaces//JointMotor.ice  (slice2cpp)
-- Adding rule to generate Laser.cpp and Laser.h from /home/lihang/robocomp//interfaces//Laser.ice  (slice2cpp)
-- Adding rule to generate RGBD.cpp and RGBD.h from /home/lihang/robocomp//interfaces//RGBD.ice  (slice2cpp)
-- Adding rule to generate TouchSensor.cpp and TouchSensor.h from /home/lihang/robocomp//interfaces//TouchSensor.ice  (slice2cpp)
-- Adding rule to generate RCISMousePicker.cpp and RCISMousePicker.h from /home/lihang/robocomp//interfaces//RCISMousePicker.ice  (slice2cpp)
-- RoboComp libraries
-- Library qmat not found in /opt/robocomp/lib
CMake Error at cmake/robocomp.cmake:62 (MESSAGE):
  Library qmat not found
Call Stack (most recent call first):
  tools/rcinnerModelSimulator/src/CMakeLists.txt:47 (ROBOCOMP_LIBRARY)

-- Configuring incomplete, errors occurred!
See also "/home/lihang/robocomp/CMakeFiles/CMakeOutput.log".
See also "/home/lihang/robocomp/CMakeFiles/CMakeError.log".
```

So the question is, is there anything missed here? I'm sure I'm following the instructions line by line.

hustcalm, 2015-03-09:
I just found that there is a `libs` directory, which has:
```bash
CMakeFiles  CMakeLists.txt  innermodel  osgviewer  qmat
```

So I try to build the libs using CMake, when running `cmake .`, I got the output:
```bash
-- The C compiler identification is GNU 4.8.2
-- The CXX compiler identification is GNU 4.8.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:3 (collect_library_directory_names):
  Unknown CMake command "collect_library_directory_names".

-- Configuring incomplete, errors occurred!
See also "/home/lihang/robocomp/libs/CMakeFiles/CMakeOutput.log".
```

I chcked my CMake version, it is:
```bash
cmake version 2.8.12.2
```

So what is the problem here?

Thanks,
Lihang

marcoag, 2015-03-09:
Hi Lihan,

Thatas a current issue, you need to install first qmat that is under
robocomp/libs/qmat as an individual project, with cmake . make and sudo
make install and then try to install the rest.

Cheers,

Marco.

-- 
Marco Antonio Gutiérrez Giraldo
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

hustcalm, 2015-03-10:
@marcoag 

Hi Marco,

Thanks for your reply. However, when running `cmake .` under `robocomp/libs/qmat`, I got those output:
```bash
-- The C compiler identification is GNU 4.8.2
-- The CXX compiler identification is GNU 4.8.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
CMake Error at CMakeLists.txt:61 (INCLUDE):
  include called with wrong number of arguments.  Include only takes one
  file.

CMake Error at CMakeLists.txt:64 (RoboComp_ADD_LIBRARY):
  Unknown CMake command "RoboComp_ADD_LIBRARY".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 2.8)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/home/lihang/robocomp/libs/qmat/CMakeFiles/CMakeOutput.log".
```

Really need to find a way to fix these issues, looking forward to your reply.

Regards,
Lihang

abhinavjain241, 2015-03-10:
Hi,

Even I am facing the same problem. Any leads on these issues?

Regards,
Abhinav

ljmanso, 2015-03-10:
I broke CMakeLists.txt a couple of days ago in an attempt to automate the instalation of the simulator. My bad.

Pull the last version of the repository. It should work now.

To install the core libraries:
cd robocomp
cmake .
make
sudo make install

Afterwards, to compile and install the simulator:
cd toolsrcinnerModelSimulator
cmake .
make
sudo make install

hustcalm, 2015-03-10:
Great, it works!!!  Thanks @ljmanso

closed 2015-03-10 by hustcalm