robocomp/robocomp #11: git annex error

nithinmurali, 2015-03-13:
running: git annex get . is giving the following error

get files/innermodel/tar36h11-0-50.png Invalid command: 'git-annex-shell 'configlist' '/~/robocomp/robocomp.git''
  You appear to be using ssh to clone a git:// URL.
  Make sure your core.gitProxy config option and the
  GIT_PROXY_COMMAND environment variable are NOT set.

  Remote origin does not have git-annex installed; setting annex-ignore
(not available) 
  No other repository is known to contain the file.
failed
get files/innermodel/tar36h11-3-50.png (not available) 
  No other repository is known to contain the file.
failed
git-annex: get: 2 failed

Lonelygeek, 2015-03-15:
same error

ljmanso, 2015-03-15:
Are you behind a proxy?

nithinmurali, 2015-03-15:
no, well we can use a proxy if we want , but i am trying this with proxy disabled.
and during cmake its giving a warning " IPPROOT environment variable not set ."

nithinmurali, 2015-03-15:
when i removed everything and did whole process again,  most of the components are getting downloaded except some
"""
get files/innermodel/tar36h11-0-50.png (not available) 
  No other repository is known to contain the file.

  (Note that these git remotes have annex-ignore set: origin)
failed
get files/innermodel/tar36h11-3-50.png (not available) 
  No other repository is known to contain the file.

  (Note that these git remotes have annex-ignore set: origin)
failed
git-annex: get: 2 failed   """

with above error

marcoag, 2015-03-20:
" IPPROOT environment variable not set ."

For this you need intel ipp (integrated performance primitves) installed and an environment variable IPPROOT pointing to the installation dir. 

Regarding git annex:
Some files on git annex might be broken, we should fix this, please list them and submit a bug report se we can work on that.

closed 2015-05-11 by ljmanso