robocomp/robocomp #13: Component Run Error

abhinavjain241, 2015-03-13:
Hey!

While following the tutorial for **"[Creating a new Component with Eclipse based Robocomp's DSL Editor](https://github.com/robocomp/robocomp/blob/master/doc/component_creation_with_DSLEditor.md)"**, I encountered the following errors after trying to run the changed component:

```sh
abhinav@abhinav-Inspiron-3521  bin/myfirstcompcomp --Ice.Config=etc/generic_config

[/home/abhinav/robocomp/classes/rapplication/rapplication.cpp]: Loading [laser:tcp -h localhost -p 10003] proxy at 'LaserProxy'...
00:06:41:667::Info::myfirstcompcomp.cpp::164::/home/abhinav/robocomp/components/mycomponents/myfirstcomp/src/myfirstcompcomp.cpp::run::LaserProxy initialized Ok!
[/home/abhinav/robocomp/classes/rapplication/rapplication.cpp]: Loading [differentialrobot:tcp -h localhost -p 10004] proxy at 'DifferentialRobotProxy'...
00:06:41:667::Info::myfirstcompcomp.cpp::175::/home/abhinav/robocomp/components/mycomponents/myfirstcomp/src/myfirstcompcomp.cpp::run::DifferentialRobotProxy initialized Ok!
00:06:41:667::Info::specificmonitor.cpp::57::/home/abhinav/robocomp/components/mycomponents/myfirstcomp/src/specificmonitor.cpp::initialize::Starting monitor ...
RoboCompMyFirstCompComp::MyFirstCompComp started
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
Network.cpp:2169: Ice::ConnectionRefusedException:
connection refused: Connection refused
```
repeatedly, and so on. I followed all the steps, ran cmake and then used KDevelop to compile and link, still it gave me this error. Any fix around this?

pbustos, 2015-03-13:
You need to have de RCIS simulator up and running,

cheers

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

abhinavjain241, 2015-03-14:
Ohh, Thank you!

closed 2015-03-22 by ljmanso

rajathkumarmp, 2015-04-02:
I am getting the same error. @pbustos could you please elaborate on how to have the RCIS simulator up and running?