robocomp/robocomp #16: git annex connection failed

DoraXingyu, 2015-03-21:
I am having trouble downloading the meshes and textures for the simulator from web using git annex.
Here's a snapshot.
I have tried  to turn my proxy off, nothing changes and my internet connection is ok.
I hope somebody can tell me how to fix this.
[image]

hustcalm, 2015-03-21:
@ljmanso @marcoag Any ideas about how to solve this problem?

Xingyu,

Hi, sorry to tell that I'm also stuck by this problem several days ago and have not got time to dig in and get it fixed.

As it complains about "Unable to establish SSL connection", so I'm thinking about see the `git annex config` to disable SSL certification or something.

See https://git-annex.branchable.com/ to try if some luck,  or do these experiments:
wget https://dl.dropboxusercontent.com/... –no-check-certificate
or
curl https://dl.dropboxusercontent.com/... --insecure

to check out if wget or curl works.

See https://fixmytechblog.wordpress.com/2013/01/31/dropbox-cant-establish-a-secure-connection/, maybe your system is blocking the Dropbox connections.

Let's fix the issue together.

Regards,
Lihang

DoraXingyu, 2015-03-21:
Lihang:
Hi, I have been trying to fix this for hours. Nothing works. Any news from your side?
I can't visit dropbox.com since the Great Fire Wall. Maybe that's the problem?
But the problem exists after I use a VPN from foreign region, although I can visit dropbox.com with Safari now.

hustcalm, 2015-03-21:
@DoraXingyu 
Bad news, I have not got the Linux box by my side for the weekend, so I will try to solve the problem by Monday maybe.

I think setting a git proxy worth trying?

Though these files are assets for running the Simulator, right? If so, maybe you can read the source code for the day or two, waiting some good news from @ljmanso @marcoag .

DoraXingyu, 2015-03-22:
@hustcalm Good news! Problem solved just now.
Something is wrong with the connection. Probably from GFW or Cernet. So I try a different VPN proxy, and it worked.
And maybe that's why you have the same problem as me since you are in Beijing too.

hustcalm, 2015-03-22:
@DoraXingyu 
Great job!!!
I will try tomorrow and let you know my situation then.

closed 2015-03-23 by DoraXingyu