robocomp/robocomp #18: cmake error

Lonelygeek, 2015-03-22:
chandy@chandy:~/robocomp$ make
Scanning dependencies of target exec_rcmanager
[  0%] Generating rcmanager
[  0%] Built target exec_rcmanager
Scanning dependencies of target ui_editorForm
[  1%] Generating ui_editorForm.py from editorForm.ui
/bin/sh: 1: pyuic4: not found
make[2]: *** [tools/rcmanager/ui_editorForm.py] Error 127
make[1]: *** [tools/rcmanager/CMakeFiles/ui_editorForm.dir/all] Error 2
make: *** [all] Error 2

rajathkumarmp, 2015-03-22:
sudo apt-get install pyqt4-dev-tools qt4-designer

The readme.md file is updated with this.

Lonelygeek, 2015-03-22:
ok is it just updated

rajathkumarmp, 2015-03-22:
Today it is updated. Anyways just install those two packages and you are good to go.

Lonelygeek, 2015-03-22:
thank u its working

closed 2015-03-22 by Lonelygeek