robocomp/robocomp #31: Error while running muecas.xml

abhi-kumar, 2015-04-04:
I got this error while running muecas.xml, I changed the library calling path too as mentioned in #25 , but still the error persists:
abhi@abhi-Inspiron-7520:~/robocomp/files/innermodel$ rcis muecas.xml---------------------------------------
--- FCL_SUPPORT: 0
---------------------------------------
Warning: STORM not found. RCISMousePicker topic will NOT be published 
InnerModelReader: reading muecas.xml
!! Saturday 04 April 2015 12:28:10.867 rcis: error: unknown exception

Please help me to get this application running.
Thanks :)

abhi-kumar, 2015-04-06:
@pbustos Could you please explain the use of STORM in all the models? I have very less knowledge regarding the use of STORM.

Thanks
Abhishek

pbustos, 2015-04-06:
Ok, I'll write I short tutorial, but ou can find a complete description of
ZeroC's Ice technology in www.zeroc.com. Being one of the best
communications middleware out there it is worth taking a good look.

Pablo

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

abhi-kumar, 2015-04-06:
@pbustos thanks for your reply. I will study it carefully and write a tutorial for robocomp tutorial module and make a pull request. If you find that useful please add it, otherwise I will make the changes people recommend and re-pull it.

abhi-kumar, 2015-04-08:
@pbustos I tried working with ice for Ubuntu 14.04, but the dependencies have broken packages for Ubuntu trusty. Do you have any other suggestion for me regarding STORM.

Regards,
Abhishek

closed 2015-05-11 by ljmanso