robocomp/robocomp #38: Installation path restricted

abhi-kumar, 2015-04-08:
Hello,

When we build the robocomp source, it is preferred, according to the documentation, that we build it in home directory. But this is restricting the installation path criteria, as we cannot build it in some other directory. Building it in some other directory will cause problems while running modules from innermodels as they call libraries from a fixed path, i.e., for e.g., "/home/robocomp/robocomp/files/osgModels/textures/wood.jpg" . Will it be helpful if we change the path in relative sense, i.e., "../osgModels/textures/wood.jpg" . Please let me know regarding the same, and I will start making the desirable changes. 

Regards,
Abhishek

nithinmurali, 2015-04-09:
abhi, actually i have included this in my GSoC proposal :-)

abhi-kumar, 2015-04-09:
@nithinmurali thats great :) . I will try and help you in this.

closed 2015-07-25 by ljmanso