robocomp/robocomp #62: Fixed errors with install.py script but outputs cmake error

rajathkumarmp, 2015-08-20:
cmake error while executing the updated install.py script

 [ Link to script ](https://github.com/rajathkumarmp/robocomp/blob/patch-2/tools/install.py)

```
Output of the script : 

﻿ln: failed to create symbolic link ‘/opt/robocomp/-1.0’: File exists
export /home/rc/robocomp >> /home/rc/.bashrc
export PATH=$PATH:/opt/robocomp/bin >> /home/rc/.bashrc
-- The C compiler identification is GNU 4.8.2
-- The CXX compiler identification is GNU 4.8.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Try OpenMP C flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Try OpenMP CXX flag = [-fopenmp]
-- Performing Test OpenMP_FLAG_DETECTED
-- Performing Test OpenMP_FLAG_DETECTED - Success
-- Found OpenMP: -fopenmp  
-- Found OpenMP
-- PYTHON BINDINGS:        NO
-- Looking for Q_WS_X11
-- Looking for Q_WS_X11 - found
-- Looking for Q_WS_WIN
-- Looking for Q_WS_WIN - not found
-- Looking for Q_WS_QWS
-- Looking for Q_WS_QWS - not found
-- Looking for Q_WS_MAC
-- Looking for Q_WS_MAC - not found
-- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- CONFIGURING PACKAGE
-- SRC_DIR is = /home/rc/robocomp
-- Version  = 1.01
-- architecture is: amd64
-- CONFIGURING DONE
-- COMPILING WITHOUT FCL SUPPORT  = 0
CMake Error at tools/rcinnerModelSimulator/src/CMakeLists.txt:46 (INCLUDE):
  include could not find load file:

    /cmake/robocompLocal.cmake

CMake Error at tools/rcinnerModelSimulator/src/CMakeLists.txt:47 (ROBOCOMP_INITIALIZE):

 Unknown CMake command "ROBOCOMP_INITIALIZE".

-- Configuring incomplete, errors occurred!
See also "/home/rc/robocomp/CMakeFiles/CMakeOutput.log".
See also "/home/rc/robocomp/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target `install'.  Stop.
/opt/robocomp/lib >> /etc/ld.so.conf
```

pbustos, 2015-08-20:
Hi Rajath,

you might want to check if the env variable ROBOCOMP is exported when you
run cmake.
Maybe insert a PRINTEV ROBOCOMP command before cmake .

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

rajathkumarmp, 2015-08-20:
@pbustos  You are absolutely right.

There is an error in this line

```
call(["echo","export", os.path.join('ROBOCOMP=',home,'robocomp'),">>",os.path.join(home,'.bashrc')])
call(["echo","export PATH=$PATH:/opt/robocomp/bin",">>",os.path.join(home,'.bashrc')])
```

rajathkumarmp, 2015-08-24:
@pbustos 
I tried the new script in my friend's laptop today and this is the error

[image]

But the same script installs everything perfectly without any error in my laptop

[image]

I am clueless as to why this is happening. Could you help me out?

pbustos, 2015-08-24:
There must be a line where de folder /opt/robocomp is created.
Insert a print statement there to check that it really exists.

Pablo

-- 
Pablo Bustos García de Castro
Laboratorio de Robótica y Visión Artificial
http://robolab.unex.es
Escuela Politécnica
Universidad de Extremadura
Avda de la Universidad s/n
Cáceres 10003
Phone: +34 927 257259
Fax: +34 927 257203

closed 2016-03-11 by ljmanso

charanjit-singh, 2019-04-08:
The command "cmake -DCMAKE_CXX_COMPILER=$COMPILER -DFCL_SUPPORT=1 . && make" exited with 1.