robocomp/robocomp #63: rcmanager stops working once it tries to ping a sleeping component

ljmanso, 2015-08-20:
rcmanager stops working once it tries to ping a sleeping component. It looks like it keeps waiting for ice-ping to return but, if any component is sleeping (e.g. it has been stopped using ctrl+z) the function does not return, hence making the tool stop working.

BasilMVarghese, 2016-04-10:
I had similar problem while running rcis launched through rcmanager(ie when we stopped rcis using ctrl+z from yakuake tab). Even we try to launch rcis again from rcmanager by clicking on up option, It was not launching.**One quick fix is down the component once.And again click on the up option**.It was because when we try to launch a sleeping component the addresses will already in use(Our sleeping rcis).So downing the sleeping component will kill it.

ljmanso, 2016-04-10:
Thank you for the feedback. Last week @rcintas started working on the issue. Instead of a single thread polling the whole component network, rcmanager will run a thread per component, so if a component is sleeping, the rest of the network will be correctly monitored.

ljmanso, 2016-04-11:
A bug fix was commited by @rcintas. It should now work properly. In case it doesn't, please, report back

closed 2016-04-11 by ljmanso