robocomp/robocomp #70: Readthedocs not working

rajathkumarmp, 2015-10-01:
@pbustos @marcoag   The build settings of readthedocs have changed and it is now building only .rst files and not .md files.

Probably a different build settings would work.

[image]

marcoag, 2016-03-23:
https://robocomp.readthedocs.org/en/latest/ is still not working

Riyuzakii, 2018-01-28:
@marcoag I would like to help fix this problem. Some advice would be appreciated.

mariabohorquez, 2018-06-12:
[https://robocomp.readthedocs.org/en/latest/](https://robocomp.readthedocs.org/en/latest/) is working now. It was a problem with the settings in ReadTheDocs. It needed to be fixed directly from there.

closed 2018-06-13 by rajathkmp