robocomp/robocomp #77: Improve response in genericworker.(py/cpp)

pbustos, 2016-08-12:
In this simple example  the program breaks without a useful message if the remote componente is not running. A previous check on the dictionary and a nice message before aborting could save many minutes.

def __init__(self, mprx):
		super(GenericWorker, self).__init__()

		self.trajectoryrobot2d_proxy = mprx["TrajectoryRobot2DProxy"]
		self.ui = Ui_guiDlg()
		self.ui.setupUi(self)
		self.show()
		self.mutex = QtCore.QMutex(QtCore.QMutex.Recursive)
		self.Period = 30
		self.timer = QtCore.QTimer(self)

GrayR00t, 2019-03-04:
I would like to do it .

pbustos, 2019-03-04:
Have you studied RoboComp and robocompdsl?

GrayR00t, 2019-03-04:
Yeah ,  I studied whatever on github if other source is there to read more
so that I would more clear to me please forward to me.

Regards
Neeraj Singh

-- 
*Disclaimer: *This email and any files transmitted with it are confidential 
and intended solely for the use of the individual or entity to whom they 
are addressed. If you have received this email in error please notify the 
system manager. This message contains confidential information and is 
intended only for the individual named. If you are not the named addressee 
you should not disseminate, distribute or copy this e-mail. Please notify 
the sender immediately by e-mail if you have received this e-mail by 
mistake and delete this e-mail from your system. If you are not the 
intended recipient you are notified that disclosing, copying, distributing 
or taking any action in reliance on the contents of this information is 
strictly prohibited.

pbustos, 2019-03-04:
Neeraj, but have in mind that this is not a GSoC topic. Topics are listed
in robocomp.org

GrayR00t, 2019-03-04:
Actually I want to contribute in gsoc idea
Learning acceptable social behaviour using machine learning technique
for this summer , please guide regarding timeline of project and exactly
what I have to do .
Regards
Neeraj Singh

-- 
*Disclaimer: *This email and any files transmitted with it are confidential 
and intended solely for the use of the individual or entity to whom they 
are addressed. If you have received this email in error please notify the 
system manager. This message contains confidential information and is 
intended only for the individual named. If you are not the named addressee 
you should not disseminate, distribute or copy this e-mail. Please notify 
the sender immediately by e-mail if you have received this e-mail by 
mistake and delete this e-mail from your system. If you are not the 
intended recipient you are notified that disclosing, copying, distributing 
or taking any action in reliance on the contents of this information is 
strictly prohibited.

pbustos, 2019-03-04:
Ok, you need to write to the mentors in charge of that topic.

Pablo

NegaNexus, 2019-04-29:
The equivalent component built in C++ doesn't have the same issue. I noticed that the C++ version of main uses uncheckedCast():
https://github.com/robocomp/robocomp/blob/e9ab83ab0be25d0d435d0d7aca3991dbf8d09c47/tools/robocompdsl/templateCPP/src/main.cpp#L27

while the Python version of main uses checkedCast() in the equivalent part of the Python template:
https://github.com/robocomp/robocomp/blob/e9ab83ab0be25d0d435d0d7aca3991dbf8d09c47/tools/robocompdsl/templatePython/src/main.py#L29

If I change the generated Python code to use uncheckedCast() instead of checkedCast(), it doesn't return the message and opens a Qt window just like the generated C++ code. I also tried the opposite in the generated C++ code, and it behaves in an analogous way to the Python code (except that the C++ component exits with a failure while the Python code continues to run even though it wasn't able to connect to the remote object).

I guess I'm curious which is the preferred behavior for this. Any thoughts?

orensbruli, 2020-03-20:
@pbustos do you know if this is still an open issue?