robocomp/robocomp #141: rcis throws memory error when run without arguements

RJ722, 2018-03-19:
When running the following command:
```
$ rcis
```

it outputs:
```
Usage: rcis InnerModelFile.xml [-p INNERMODEL_MANAGER_PORT] [-f MSECS]
[1]    3346 abort (core dumped)  rcis
```

Since we are getting "core dumped", I suspect that we are not terminating the program properly as it is trying to read from a pointer which might not exist at that time. Is this intentional? If not, can we handle this more gracefully?

marcoag, 2018-05-05:
You're probably missing the 3D model files referenced inside the xml file. Please check you have them all.

RJ722, 2018-05-08:
Yes, that was intentional. I am just saying that wouldn't it be better if we could handle this situation more gracefully, by perhaps showing an error message rather than crashing the program.

bartekxk, 2019-04-07:
I added the default parameter in rcis, I think this can be a good solution. #196

orensbruli, 2021-02-19:
Fixed by @bartekxk https://github.com/robocomp/robocomp/pull/196

closed 2021-02-19 by orensbruli