robocomp/robocomp #143: Error in git annex get .

coolprinshu123, 2018-03-23:
When I am doing 
$git annex get .
it is showing me errors.

marcoag, 2018-05-05:
can you please post the actual errors? what fixing did you do?

NegaNexus, 2019-03-28:
Since this is still an open issue I'll also say that I'm getting the same issue (I assume, anyway)

```
$ git annex get .
get files/osgModels/Adapta/Airport/gallery.ive (from web...) 
/home/thys/robocomp/.git/anne 100%[================================================>] 181.13K   682KB/s    in 0.3s    
2019-03-28 01:24:54 URL:https://dl.dropboxusercontent.com/s/umzxfd8my6kacq8/panel.ive?dl=1 [185479/185479] -> "/home/thys/robocomp/.git/annex/tmp/SHA256E-s945490--499f34ab58e7465dfa7cde77b1365102f945c6c182d7a0f3c372475eb7c48e94.ive" [1]

  verification of content failed

  Unable to access these remotes: web

  Try making some of these repositories available:
  	00000000-0000-0000-0000-000000000001 -- web
   	03a7f2f8-3678-46ea-98e3-4e1c2ca94f68 -- daniel@Daniel-PC:~/robocomp
   	15589b00-3673-4368-bdd9-555b3844fa45 -- spyke@deepblue:~/robocomp
   	19cac3dd-ff66-4d4c-9b62-a5d617a3f244 -- mario@majendi:~/robocomp
   	2e65c1a8-40ce-42e3-a960-831c00cc0ba4 -- daniel@salabeta4:~/robocomp
   	505cfea7-2900-4a0c-b85a-156b98b182a2 -- spyke@silicon:~/robocomp
   	5481651d-e37e-49f1-916c-0305a81798e8 -- mercedes@lmanso:~/robocomp
   	54f91217-72a0-4934-8b0d-686ab56f040a -- computaex@computaex:~/robocomp
   	566b3e09-9d88-470c-838e-905de3eaa107 -- robolab@robonuc1:~/robocomp
   	636e4afd-ef4f-4156-bb79-3731b0dbb37e -- ljmanso@negrino:~/robocomp
   	759fea42-078f-11e4-842f-4bf67ca95a49
   	85b71dc1-5958-4d6e-ad09-0a05bd419083 -- spyke@deepblue:~/robocomp
   	8972ec57-8799-4954-bf6f-24258388feca -- salabeta@salabeta4:~/robocomp
   	963b5d6e-dc3d-402a-ac39-a69b3a22968d -- luiky@luiky:~/robocomp
   	969dc093-01d4-49ef-8951-9b0ee50f7941 -- salabeta8@salabeta1:~/robocomp
   	9bed57a2-af25-4070-b873-a5ccf7238145 -- ljmanso@ljmansolab:~/robocomp
   	a3d02b20-f041-4678-8473-3e6027551666 -- pbustos@robex:~/robocomp
   	afe6f6ca-b737-418b-b7cb-3f2c54e9a2f1 -- ivan@ivan-Aspire-5742G:~/robocomp
   	b8580506-f86a-11e3-9ded-0bae4cd33c4a -- felipe@felipe-System-Product-Name:~/robocomp
   	b9414698-b998-47c0-8a34-c5c21a3dece2 -- agustin@lrva:~/robocomp
   	e3d92629-b3ef-4e86-a132-1fcdead5e008 -- robolab@gualzru5:~/robocomp
   	e8680ea1-d119-4cc2-a1b0-9a1213a240e3 -- jose@jose-K55VD:~/robocomp
   	f98731df-a20a-4300-b1d3-74d55a62e887 -- marcog@thinkie:~/robocomp

  (Note that these git remotes have annex-ignore set: origin)
failed
get files/osgModels/Adapta/Airport/panel.ive (from web...) 
/home/thys/robocomp/.git/anne 100%[================================================>] 181.13K   577KB/s    in 0.3s    
2019-03-28 01:24:56 URL:https://dl.dropboxusercontent.com/s/umzxfd8my6kacq8/panel.ive?dl=1 [185479/185479] -> "/home/thys/robocomp/.git/annex/tmp/SHA256E-s945490--499f34ab58e7465dfa7cde77b1365102f945c6c182d7a0f3c372475eb7c48e94.ive" [1]

  verification of content failed

  Unable to access these remotes: web

  Try making some of these repositories available:
  	00000000-0000-0000-0000-000000000001 -- web
   	03a7f2f8-3678-46ea-98e3-4e1c2ca94f68 -- daniel@Daniel-PC:~/robocomp
   	15589b00-3673-4368-bdd9-555b3844fa45 -- spyke@deepblue:~/robocomp
   	19cac3dd-ff66-4d4c-9b62-a5d617a3f244 -- mario@majendi:~/robocomp
   	2e65c1a8-40ce-42e3-a960-831c00cc0ba4 -- daniel@salabeta4:~/robocomp
   	505cfea7-2900-4a0c-b85a-156b98b182a2 -- spyke@silicon:~/robocomp
   	5481651d-e37e-49f1-916c-0305a81798e8 -- mercedes@lmanso:~/robocomp
   	54f91217-72a0-4934-8b0d-686ab56f040a -- computaex@computaex:~/robocomp
   	566b3e09-9d88-470c-838e-905de3eaa107 -- robolab@robonuc1:~/robocomp
   	636e4afd-ef4f-4156-bb79-3731b0dbb37e -- ljmanso@negrino:~/robocomp
   	759fea42-078f-11e4-842f-4bf67ca95a49
   	85b71dc1-5958-4d6e-ad09-0a05bd419083 -- spyke@deepblue:~/robocomp
   	8972ec57-8799-4954-bf6f-24258388feca -- salabeta@salabeta4:~/robocomp
   	963b5d6e-dc3d-402a-ac39-a69b3a22968d -- luiky@luiky:~/robocomp
   	969dc093-01d4-49ef-8951-9b0ee50f7941 -- salabeta8@salabeta1:~/robocomp
   	9bed57a2-af25-4070-b873-a5ccf7238145 -- ljmanso@ljmansolab:~/robocomp
   	a3d02b20-f041-4678-8473-3e6027551666 -- pbustos@robex:~/robocomp
   	afe6f6ca-b737-418b-b7cb-3f2c54e9a2f1 -- ivan@ivan-Aspire-5742G:~/robocomp
   	b8580506-f86a-11e3-9ded-0bae4cd33c4a -- felipe@felipe-System-Product-Name:~/robocomp
   	b9414698-b998-47c0-8a34-c5c21a3dece2 -- agustin@lrva:~/robocomp
   	e3d92629-b3ef-4e86-a132-1fcdead5e008 -- robolab@gualzru5:~/robocomp
   	e8680ea1-d119-4cc2-a1b0-9a1213a240e3 -- jose@jose-K55VD:~/robocomp
[... 15 lines cut ...]
   	f98731df-a20a-4300-b1d3-74d55a62e887 -- marcog@thinkie:~/robocomp

  (Note that these git remotes have annex-ignore set: origin)
failed
get files/osgModels/robexarm/barracolumna.ive (from web...) 
/home/thys/robocomp/.git/anne     [  <=>                                            ] 183.39K   617KB/s    in 0.3s    
2019-03-28 01:24:58 URL:https://www.dropbox.com/s/zfof39zqeyaj4ci/barracolumna.ive [187787] -> "/home/thys/robocomp/.git/annex/tmp/SHA256E-s133000--a6c7403f869f6837eaa7bc62d453ae4996fb9c2679e1e1abb1cd16284df45445.ive" [1]

  verification of content failed

  Unable to access these remotes: web

  Try making some of these repositories available:
  	00000000-0000-0000-0000-000000000001 -- web
   	03a7f2f8-3678-46ea-98e3-4e1c2ca94f68 -- daniel@Daniel-PC:~/robocomp
   	19cac3dd-ff66-4d4c-9b62-a5d617a3f244 -- mario@majendi:~/robocomp
   	505cfea7-2900-4a0c-b85a-156b98b182a2 -- spyke@silicon:~/robocomp
   	963b5d6e-dc3d-402a-ac39-a69b3a22968d -- luiky@luiky:~/robocomp
   	afe6f6ca-b737-418b-b7cb-3f2c54e9a2f1 -- ivan@ivan-Aspire-5742G:~/robocomp
   	f98731df-a20a-4300-b1d3-74d55a62e887 -- marcog@thinkie:~/robocomp

  (Note that these git remotes have annex-ignore set: origin)
failed
git-annex: get: 4 failed
```

Goshikhar23, 2020-03-03:
Make sure you have an active internet connection while using git annex.

orensbruli, 2020-03-03:
Removed old deprecated files from git annex.

closed 2020-03-03 by orensbruli