robocomp/robocomp #200: Robolab tool

bartekxk, 2019-04-11:
Hello,
Would it be a good idea to create a tool for easy component run? I thought about doing it based on the environmental variable, but I do not know if this idea matches the idea of components.
For example, we could run anywhere to start keyboardrobotcontroller component with defualt config file:
`robolab keyboardrobotcontroller`
We could add parameters to the run with installation or set other config file path.

orensbruli, 2019-04-12:
Hello,
I think this old tool had the porpose that you are suggesting to implement:
https://github.com/robocomp/robocomp/blob/stable/tools/buildTools/rcrun.py
But as you have said with a component based architecture this tools are currently not used by robocomp users. Tthe development is currently focused on rcmanager (as a replacemente of rcmanagersimple) to deploy large numbers of components on local and remote machines.
https://github.com/robocomp/robocomp/tree/development/tools/rcmanagerSimple
https://github.com/robocomp/robocomp/tree/development/tools/rcmanager
Best.

bartekxk, 2019-04-12:
That's right, it's better done. I'm closing this isuee.
Best.

closed 2019-04-12 by bartekxk