robocomp/robocomp #206: Missing Code Linter and Build scripts before commit.

charanjit-singh, 2019-04-20:
This project is missing the code linting scripts, i.e. before pushing the code. and also automatic tests scripts on local system.

charanjit-singh, 2019-04-20:
Can I work on this issue?

orensbruli, 2019-04-22:
I'm interested to know what you have in mind. How do you want to implement it?

charanjit-singh, 2019-04-23:
We can run some scripts before making a commit to format the code.
https://githooks.com/

bartekxk, 2019-04-24:
The project compilation is tested by travis. I think it is impossible to enter automatic tests into the entire project, it is too large. Only for specific tools. I do not understand it very much, but if you have an idea and you could describe it better, I will be happy to help you.

orensbruli, 2019-04-24:
@bartekxk Travis is not working because they don't support 18.04 and robocomp currently need it.
@charanjit-singh As @bartekxk suggest, the core of Robocomp is pretty is big and maybe it's difficult to currently make it converge to some code styling (but it would be great). Also there's a mix of python and c++ code.
Maybe, if you can show some output of this scripts that you suggests executed over robocomp core source would give us a good idea of how it can contribute to a better coding practices.

closed 2021-02-19 by orensbruli