robocomp/robocomp #207: Creating a  gentle introduction to RoboComp

sunn-e, 2019-05-04:
Hi, My name is Sunny. I'm a computer science student. I want to create an user guide as described here https://robocomp.github.io/web/gsod/2019/ideas/
@pbustos @pilarbachiller How do I start contributing?

sunn-e, 2019-05-04:
Is there something I should read first.

pbustos, 2019-05-04:
Hi Sunny,

thanks for your interest. You can take a good look at the project by
cloning RoboComp and going through the tutorials. As you will see, this
project is to make this process easier for beginners. Now it  has a more
technical, "just what is needed to go" organization.

Pablo

pbustos, 2019-05-27:
Hello Sunny, please take a look at the gitter channel. We are going to post
there a small assignment to check the documentation skills of the
candidates.

robocomp/GSoD

Pablo

sunn-e, 2019-05-28:
@pbustos Thanks for the reply. I'll go and check it right now.

sunn-e, 2019-05-28:
Am I suppose to go to the GSoD channel( https://gitter.im/robocomp/GSoD?source=orgpage) or the refular robocomp channel.

pbustos, 2019-05-28:
The GSoD channel.

sunn-e, 2019-05-28:
Okay. Thanks. When will that be posted?

pbustos, 2019-05-29:
We are working on it. Very soon.

sunn-e, 2019-05-30:
Okay. Please let me know once you are done with preparations. :)

pbustos, 2019-06-03:
They are ready in the gitter channel

sunn-e, 2019-06-03:
I know, I was the first one to comment on it. ;) Thanks for the reminder.

closed 2020-12-22 by orensbruli