robocomp/robocomp #217: robocompdsl have no output if a file with invalid extension is given

orensbruli, 2019-06-27:
robocompdsl have no output if a file with invalid extension is given.
Some error should be shown

niveditarufus, 2020-01-16:
@orensbruli  I have added a line which displays an error message when a file with invalid extension is given.

orensbruli, 2020-03-19:
Sorry @niveditarufus, where did you add the line? Did you make a Pull Request?

niveditarufus, 2020-03-19:
Yeah, I did make a pull request when the issue was still open and the before the codebase was updated.....Now I think it does display an error message upon given wrong or no extension and the pull request was closed, so no worries.

rahulkatiyar19955, 2020-04-02:
if we try to execute the following cmd with the wrong file, No error message is shown on the terminal.
`robocompdsl new_component.cdsla .`

Note this should say something like 

> Please check the Input file 
Input File should be either .cdsl or .idsl

closed 2020-12-22 by orensbruli