robocomp/robocomp #226: Remove support for PyParsing 2.2 [to be implemented in October 2020]

ljmanso, 2019-10-06:
The structure resulting from parsing CDSL, IDSL and SMDSL files changes from PyParsing 2.2 to PyParsing 2.4. Until the introduction of a couple of exception handlers in the corresponding code, this resulted in errors when generating components if PyParsing 2.4 was installed.

Although it is necessary to have these exception handlers for as long as we have users with PyParsing 2.2, we estimate that all users will have migrated to 2.4 by October 2020. Therefore, in October 2020 we should remove the exception handlers and assume that users will have PyParsing 2.4. This implies that we should remove the code in the "try" and the code in the "except" block should remain.

The exception handlers have been marked with the following comment:
"Horrible hack to make robocompdsl work with pyparsing > 2.2".

orensbruli, 2020-12-23:
Refactored.

orensbruli, 2020-12-28:
Reverted. Pyparsing 2.2 is the default python3 pyparsing package in Ubuntu 18.04 (https://launchpad.net/ubuntu/bionic/+package/python3-pyparsing). This needs to be kept until 18.04 is no longer supported by Robocomp.