robocomp/robocomp #242: Add some info in STUBS about the Interface

pbustos, 2020-02-24:
For example, if a component subscribes to JoystickAdapter.idsl, generate:

        #
	#  SUBSCRIPTON to sendData method from JoystickAdapter interface
	#
	def JoystickAdapter_sendData(self, data):
		#
		#subscribesToCODE
		#
		pass

rahulkatiyar19955, 2020-02-27:
@pbustos I worked on this issue and found that a small change in "robocomp/tools/robocompdsl/templatePython/src" directory with the filename specificworker.py at line no. 204 was required.
[https://i.imgur.com/THBqpIy.png?1](https://i.imgur.com/THBqpIy.png?1)
Can you suggest me how to commit to the development branch?

orensbruli, 2020-02-27:
@rahulkatiyar19955 
Fork the robocomp repository.
Clone it.
Do the changes that you think you have to do. Make the commits and push as usual.
When you have finished, open a Pull Request.

https://help.github.com/en/github/getting-started-with-github/fork-a-repo

kushagra414, 2020-03-19:
@jpbandera please close this issue if it is resolved.

closed 2020-03-19 by orensbruli