robocomp/robocomp #247: Main.cpp segfaults if IceStorm is not in config file.

pbustos, 2020-03-06:
If this line is not in the config file and the component publishes or subscribes to something, main.cpp segfaults without further information.

# This property is used by the clients to connect to IceStorm.
TopicManager.Proxy=IceStorm/TopicManager:default -p 9999

orensbruli, 2020-03-18:
Hello Pablo,
Next time, it's easier to have different issues open for different problems.

For your last comment:
> Also, when regenerating a component and including a subscribe entry in its cdsl, the TopicManager line is not included in the new config file.

I can't reproduce it. 
I get this changes on the config file:
[image]

When I add the subscribesTo to the cdsl:
[image]

orensbruli, 2020-03-18:
About the first issue you describe:
> If this line is not in the config file and the component publishes or subscribes to something, main.cpp segfaults without further information.
> 
> // This property is used by the clients to connect to IceStorm.
> TopicManager.Proxy=IceStorm/TopicManager:default -p 9999

I think it's been fixed with the commit https://github.com/robocomp/robocomp/commit/7c31593501339de1116cfb774344c9fa1c11e226
so I proceed to close this issue. If you can reproduce the second problem, please open a new issue.

It would be great if you can also check it.
Best.

closed 2020-03-18 by orensbruli