robocomp/robocomp #248: Problem while installing Robocomp due to Anaconda

kushagra414, 2020-03-07:
The `robocompdsl` command uses the python path specified by **Anaconda**.
- Check all the python paths installed on the system.
`type -a python`
[image]
- Verify which python is being used by default by using the command
`which python`<br>
[image]<br>
To get rid of this problem we need to comment out the anaconda path in `.bashrc` file under `home` directory. The code in comments should be indented like shown in screenshot<br>
**[image]**<br>
- After saving the changes. Open a new terminal and reload all bashrc commands using
`source ~/.bashrc`

GarvTambi, 2020-03-11:
I faced this issue but because of this, I resolve it out...
Thanks @kushagra414

orensbruli, 2020-03-18:
Thank you for your contribution @kushagra414

closed 2020-03-18 by orensbruli