robocomp/robocomp #257: The ling of installation script is broken [BUG]

hugoziviani, 2020-03-20:
The link of quick installation of robocomp is broken.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to: https://github.com/robocomp/robocomp/blob/development/doc/FAQ.md
2. On "How should I download Robocomp?" 
3. Click on the "the installation script"
The link is broken.

**Expected behavior**
The page with the script cloud open

This is the broken link:
https://github.com/robocomp/robocomp/blob/development/InstallScript.md

orensbruli, 2020-03-20:
Fixed with https://github.com/robocomp/robocomp/commit/0d01f0dfd4492a5f0629d3a031064f6d5387ffec
https://github.com/robocomp/robocomp/commit/e54e93fc64665b5702a7ef33546c5015ffeb06cc
and
https://github.com/robocomp/robocomp/commit/daeb184e4bff47f5e4f6ab970e9e6d05966c6d41

closed 2020-03-20 by orensbruli

hugoziviani, 2020-03-20:
Grate! You're fast! cheers.