robocomp/robocomp #272: DSR Auxiliary sub-APIs can't be instantiated from G

DarkGeekMS, 2020-08-17:
DSR Auxiliary sub-APIs (RT & IO) can't be instantiated from G, as mentioned [here](https://github.com/robocomp/robocomp/tree/development/classes/dsr/#Auxiliary-sub-APIs). Only Innermodel sub-API can.

However, I can call their methods directly from Core API (not through the sub-API) , like :
```cpp
G->write_to_json_file("./"+agent_name+".json");
G->insert_or_assign_edge_RT(node, id, trans, rot);
```

I think this should be clarified in the documentation or fixed in the API itself.

orensbruli, 2020-08-18:
Yes, thank you. As you have discovered, I think this was a good intention more than a current implementation but as you say it would be clarified in the documentation or implemented in the API. 
If you have some time we could work on a proposal for those sub-APIs to be implemented and accessed in a way similar to Innermodel one...

DarkGeekMS, 2020-08-18:
Sure Thing, I'd be more than happy to work on that.

closed 2020-12-22 by orensbruli