robocomp/robocomp #277: [BUG] Link error in new DSR components compilation

DarkGeekMS, 2020-09-02:
**Description**
When creating a new component that includes `options dsr;`, the component compilation fails to link with most DSR classes. This is because not all headers and `cpp` sources are included in `CMakeListsSpecific.txt`.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a `.cdsl` file with `options dsr;`.
2. Generate the component code.
3. `cmake . && make`
4. The errors appear, while linking.

**Suggeted Solution**
I think this should be fixed in the code generator. To include all DSR headers and `cpp` sources in `CMakeListsSpecific.txt`, like so :
```
# Sources set
SET ( SOURCES
  specificworker.cpp
  specificmonitor.cpp
   $ENV{ROBOCOMP}/classes/dsr/api/dsr_api.cpp
   $ENV{ROBOCOMP}/classes/dsr/core/types/crdt_types.cpp
   $ENV{ROBOCOMP}/classes/dsr/core/types/user_types.cpp
   $ENV{ROBOCOMP}/classes/dsr/core/rtps/dsrpublisher.cpp
   $ENV{ROBOCOMP}/classes/dsr/core/rtps/dsrsubscriber.cpp
   $ENV{ROBOCOMP}/classes/dsr/core/rtps/dsrparticipant.cpp
   $ENV{ROBOCOMP}/classes/dsr/core/topics/IDLGraphPubSubTypes.cxx
   $ENV{ROBOCOMP}/classes/dsr/core/topics/IDLGraph.cxx
   $ENV{ROBOCOMP}/classes/dsr/api/dsr_inner_api.cpp
   $ENV{ROBOCOMP}/classes/dsr/api/dsr_utils.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/dsr_gui.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/osg_3d_viewer/osg_3d_viewer.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/qscene_2d_viewer/qscene_2d_viewer.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/graph_viewer/graph_viewer.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/graph_viewer/graph_node.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/graph_viewer/graph_edge.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/tree_viewer/tree_viewer.cpp
   $ENV{ROBOCOMP}/classes/dsr/gui/viewers/_abstract_graphic_view.cpp
)
```
```
# Headers set
SET ( HEADERS
  specificworker.h
  specificmonitor.h
   $ENV{ROBOCOMP}/classes/dsr/api/dsr_api.h
   $ENV{ROBOCOMP}/classes/dsr/gui/dsr_gui.h
   $ENV{ROBOCOMP}/classes/graph-related-classes/CRDT.h
   $ENV{ROBOCOMP}/classes/graph-related-classes/CRDT_graphviewer.h
   $ENV{ROBOCOMP}/classes/graph-related-classes/CRDT_graph.h
   $ENV{ROBOCOMP}/classes/graph-related-classes/CRDT_graphnode.h
   $ENV{ROBOCOMP}/classes/graph-related-classes/CRDT_graphedge.h
)
```

rcintas, 2020-09-02:
A couple files were missing, 

Please check if issue has been solved

DarkGeekMS, 2020-09-02:
Yes, the issue is now resolved.

closed 2020-09-02 by DarkGeekMS