robocomp/robocomp #286: Add workspace.py to python path

pawanw17, 2020-12-31:
Adding workspace.py to pythonpath will help remove the line
 ``
import sys
sys.path.append('/opt/robocomp/python')
``
from build tools

orensbruli, 2021-01-04:
Can you suggest a change to the installation scripts or documentation to get it done?
I think it's better to have it in the code that expecting people to manually update their python path. 
But maybe we can find an automated way to get this done?

pawanw17, 2021-01-04:
we can add 
``
export PYTHONPATH=/opt/robocomp/python:$PYTHONPATH
``
in .bashrc 
and in the installation instructions add this line, but I guess the method already implimented is on the safer side.

orensbruli, 2021-01-05:
Maybe this could be a solution to be evaluated?
https://medium.com/@arnaud.bertrand/modifying-python-s-search-path-with-pth-files-2a41a4143574
Creating a .pth file pointing to /opt/robocomp/python and install the .pth file on robocomp installation?

pawanw17, 2021-01-06:
I tested this and can confirm that this solution works well without adding an extra installation instruction.

closed 2021-01-09 by pawanw17