robocomp/robocomp #287: [FEATURE] Add autocomplete to rccd.

pawanw17, 2021-01-04:
**Is your feature request related to a problem? Please describe.**
rccd is a very useful tool to redirect to various components quickly, but there is no autocomplete feature.

**Describe the solution you'd like**
Ability to autocomplete the component names in order to quickly switch between components.

orensbruli, 2021-01-04:
It would be great if you can look into this!
But I guess there's a big problem with this and it's that you can't change the directory of the current session within a python script. To get this working we use a mix of sh and python and I'm not sure if it would be easy to get autocompletion for the sh code.

https://github.com/robocomp/robocomp/blob/development/tools/buildTools/rccd_utils.py
https://github.com/robocomp/robocomp/blob/development/tools/buildTools/rccd.sh
https://github.com/robocomp/robocomp/blob/development/tools/buildTools/setup.sh#L3

pawanw17, 2021-01-04:
Yes I saw these files before, made this issue so If anyone interested in robocomp can get started, I'll see if I can do something about it.

Arnab11917676, 2021-01-28:
Hello sir, I want to work on this issue. It would be great if you will allow me to do so.
Thank you.

orensbruli, 2021-01-28:
Hello @Arnab11917676
Take a look at the related PR to this issue and we can discuss there if you can contribute in any way.
https://github.com/robocomp/robocomp/pull/294
Thank you!

closed 2021-02-16 by pawanw17