robocomp/robocomp #289: [BUG]

pawanw17, 2021-01-08:
### Describe the bug
The link to google groups in the README.md is broken. 

### Instructions
This is a ``first-timer-ony`` issue. **So the people who have not contributed to robocomp's codebase before, or people who have not contributed to open source before can take this up.**

####Prerequisites
You do not need to know anything prior to taking up this issue.

####Steps
- [ ] **Claim this issue** : Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!
- [ ]  **Fork the robocomp/robocomp repository** using the fork button on the topright corner.
- [ ] **Clone the newly forked repository** on your local machine.
- [ ] **Update** the file README.md by removing the broken link.
- [ ] **Commit** your changes.
- [ ] **Make a Pull Request** by going to your fork of robocomp.

If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow on how GitHub works.](https://guides.github.com/activities/hello-world/)

If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial on how to send a pull request using the terminal.](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github)

###Questions?
Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below or ask in the [gitter channel](https://gitter.im/robocomp).

Samiatrix, 2021-01-12:
I would like to claim this issue.

orensbruli, 2021-01-12:
Perfect. Go for it. Welcome!

closed 2021-01-12 by pawanw17