robocomp/robocomp #297: Complete DSR start tutorial with screenshots or videos.

orensbruli, 2021-01-25:
### Describe the bug

Add screenshots or video for the steps of this tutorial:
https://github.com/robocomp/robocomp/blob/development/doc/DSR-start.md

### Instructions
This is a ``first-timer-only`` issue. **So the people who have not contributed to robocomp's codebase before, or people who have not contributed to open source before can take this up.**

If you have contributed before, **consider leaving this PR for someone new** :)

#### Prerequisites
You do not need to know anything prior to taking up this issue! Welcome to opensource!

#### 📋 Steps
- [ ] **Claim this issue** : Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ]  **Fork the robocomp/robocomp repository** using the fork button on the topright corner.

- [ ] **Clone the newly forked repository** on your local machine.

- [ ] **Update** the file README.md by removing the broken link.

- [ ] **Commit and push** your changes.

- [ ] **Make a Pull Request** by going to your fork of robocomp.

- [ ] **Ask in comments for a maintainer to review**.

If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow on how GitHub works.](https://guides.github.com/activities/hello-world/)

If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial on how to send a pull request using the terminal.](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github)

### Questions?
Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below or ask in the [gitter channel](https://gitter.im/robocomp).

[Robocomp FAQ](https://github.com/robocomp/robocomp/blob/development/doc/FAQ.md)
[Robocomp Tutorials](https://github.com/robocomp/robocomp/blob/development/doc/README.md)
[Robocomp Contribution guide](https://github.com/robocomp/robocomp/blob/development/CONTRIBUTING.md)

**Good luck with your first issue!**

Sirish07, 2021-01-30:
@orensbruli Thanks. I am new to the community and this looks like a good starting point. Will send PR at the earliest.

Ddevendra, 2021-02-18:
Hey, I am new to the community and was looking at this [newcomers] issue 
- I have successfully installed Robocomp from source 
- I was following this tutorial (https://github.com/robocomp/robocomp/blob/development/doc/DSR-start.md) 
and got stuck at running "bash run.sh" in directory "components/dsr-graph/robots_pyrep/viriatoPyrep"
[image]

pawanw17, 2021-02-19:
Hello @Ddevendra before running the component you first have to build it, which can be done by going to the component directory and running:
```
cmake .
make 
```

Ddevendra, 2021-02-21:
#Yes!! It Worked

Now I have successfully run the "DSR-start.md" tutorial.

I have following suggestions that can be done to this tutorial guide: 

- coppeliaSim, pyrep installation missing (adding instructions, links for that)
- missing commands for setting environment variables (like ROBOCOMP and COPPELIASIM_ROOT)
- add images/button-icons/snap for copelliaSim and graph screens
- rearrange commands with respective instruction (some of the commands are stated after the set of instructions, commands can be placed right next to its instruction )
- a troubleshooting section can also be added at the end to address some commonly encountered errors

If someone can check/confirm the suggestions, I will start working on it and make a PR when its all done

pbustos, 2021-02-21:
Hello Devendra. Thanks for the suggestions. Please go ahead with the PR.

Pablo

Ddevendra, 2021-02-25:
There is already a PR pending ["minor fixes in DSR_start.md" ](https://github.com/robocomp/robocomp/pull/308/commits/c5ff253b1959d41789f8671c2dcc8b81e1ac777d)

should I make changes to the current DSR_start.md file and submit a PR ?

pbustos, 2021-02-25:
Yes, please!

Ddevendra, 2021-02-25:
You can compare the updated document "DSR_start.md" from [here](https://github.com/Ddevendra/robocomp/blob/development/doc/DSR-start.md)

pbustos, 2021-02-26:
It is just great. I accepted the PR. Do you need anything else?

We are thinking of creating a second version of this tutorial, slightly
more advanced. Changes in G automatically produce Qt signals and the code
generator creates empty slots for them. So the next version would fill one
of these slots for the laser data, and when data arrives it will store it
in a DoubleBuffer thread-safe store and when the compute loop asks for
available data, it will receive the processed result (done inside
DoubleBuffer) of the distance calculation.
The idea is that all processing in CORTEX are transformations of G
elements, except for ViriatoPyrep that acts as a gateway with the body. You
can see them as processing fibers that read and write in G, creating,
deleting or editing nodes. Of course, these dynamics respond to several
goals, internal and external. In the first example, the goal is maintained
by the new agent as a move and not collide behaviour.

Pablo

Ddevendra, 2021-02-27:
Hello Pablo, Thanks!
I am very much looking forward to contribute to robocomp as I am considering to participate in GSoC 2021.
I would love to work on other issues but as I am a newbie, I don't have much understanding about robocomp. (I didn't understand the above described idea)

I guess I will complete the tutorials (as mentioned on GSoC-2021 FAQs) and get back to you

pbustos, 2021-02-27:
Sure, don't worry. You'll get to it.
Ask any questions you need.

Pablo

closed 2021-03-14 by orensbruli