robocomp/robocomp #304: [BUG] Installation of pybind11 from ppa leads to compilation issues

pawanw17, 2021-01-30:
### Describe the bug
Installation of pybind11 form ppa as described here https://github.com/robocomp/robocomp/tree/development/classes/dsr 
leads to compilation issues as no pybind11/iostream.h is generated.
```
fatal error: pybind11/iostream.h: No such file or directory
   14 | #include <pybind11/iostream.h>
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
python-wrapper/CMakeFiles/pydsr.dir/build.make:94: recipe for target 'python-wrapper/CMakeFiles/pydsr.dir/python_api.cpp.o' failed
make[2]: *** [python-wrapper/CMakeFiles/pydsr.dir/python_api.cpp.o] Error 1
CMakeFiles/Makefile2:344: recipe for target 'python-wrapper/CMakeFiles/pydsr.dir/all' failed
make[1]: *** [python-wrapper/CMakeFiles/pydsr.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

However installation from source works fine
### Your environment
 - Ubuntu/Linux version: [e.g. Ubuntu 18.04]
 - Python / GCC Version: [e.g. Python 3.6.9]
   
 - [x] DSR dependencies have been installed.  

### To Reproduce
Steps to reproduce the behavior:
1. cd '~/robocomp/classes/dsr/'
2. mkdir build 
3. cd build
4. cmake ..
5. make -j$(nproc)
6. see the error

### If the bug is confirmed, would you be willing to submit a PR?
Yes

orensbruli, 2021-03-14:
Thank you!
Can I guess for the change on the title that you have tested this in bionic and focal?

orensbruli, 2021-03-14:
I can't reproduce this error in the Docker focal image of robocomp. Could you check too?

pawanw17, 2021-03-14:
Hey @orensbruli PPA for pybind11 works in focal, this error is specific to bionic but as DSR doesn't support bionic yet so should I close this issue?

orensbruli, 2021-03-14:
I think we can.
If we find what's is the difference we could try to suggest a fix in https://github.com/pybind/pybind11 ?

pawanw17, 2021-03-14:
The only difference is of the file pybind/iostream.h which is installed in focal but not in bionic.

orensbruli, 2021-03-14:
Getting some more information:
robolab@aa902e13c721:~/robocomp$ dpkg -S /usr/include/pybind11
pybind11-dev: /usr/include/pybind11

orensbruli, 2021-03-14:
So, my doubt now is... iostream wasn't in the bionic package because it didn't exist or because of a "bug"?

orensbruli, 2021-03-14:
I need confirmation, but I think we could remove those imports from our code.
@JuanCarlosgg can you confirm if pybind/iostream.h is really used or not?
```bash
robolab@robolab-server$: grep -r pybind11/iostream
python-wrapper/signal_function_caster.h:#include <pybind11/iostream.h>
python-wrapper/python_api.cpp:#include <pybind11/iostream.h>
```

closed 2021-04-27 by orensbruli