robocomp/robocomp #311: [Newcomers issue] Broken link

pawanw17, 2021-02-22:
### Describe the bug

A link in [first_issue.md](https://github.com/robocomp/robocomp/blob/development/.github/ISSUE_TEMPLATE/first_issue.md) is broken.

### Instructions
This is a ``first-timer-only`` issue. **So the people who have not contributed to robocomp's codebase before, or people who have not contributed to open source before can take this up.**

If you have contributed before, **consider leaving this PR for someone new** :)

The contributor has to replace a broken link
```diff
diff --git a/.github/ISSUE_TEMPLATE/first_issue.md b/.github/ISSUE_TEMPLATE/first_issue.md
index 2a8db59a..d3f54b86 100644
--- a/.github/ISSUE_TEMPLATE/first_issue.md
+++ b/.github/ISSUE_TEMPLATE/first_issue.md
@@ -35,7 +35,7 @@ You do not need to know anything prior to taking up this issue! Welcome to opens
 
 If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow on how GitHub works.](https://guides.github.com/activities/hello-world/)
 
-If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial on how to send a pull request using the terminal.](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github)
+If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial on how to send a pull request using the terminal.](https://github.com/robocomp/robocomp/blob/development/doc/contribute/contribute.md)
 ```
 ### Questions?
 Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below or ask in the [gitter channel](https://gitter.im/robocomp).

#### Prerequisites
You do not need to know anything prior to taking up this issue! Welcome to opensource!
Use this great [tutorial](https://github.com/robocomp/robocomp/blob/development/doc/contribute/contribute.md) if you have never contributed before.

#### 📋 Steps
- [ ] **Claim this issue** : Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ]  **Fork the robocomp/robocomp repository** using the fork button on the topright corner.

- [ ] **Clone the newly forked repository** on your local machine.

- [ ] **Update** the file [first_issue.md](https://github.com/robocomp/robocomp/blob/development/.github/ISSUE_TEMPLATE/first_issue.md) by replacing the broken link with https://github.com/robocomp/robocomp/blob/development/doc/contribute/contribute.md

- [ ] **Commit and push** your changes.

- [ ] **Make a Pull Request** by going to your fork of robocomp.

- [ ] **Ask in comments for a maintainer to review**.

If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow on how GitHub works.](https://guides.github.com/activities/hello-world/)

### Questions?
Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below or ask in the [gitter channel](https://gitter.im/robocomp).

[Robocomp FAQ](https://github.com/robocomp/robocomp/blob/development/doc/FAQ.md)
[Robocomp Tutorials](https://github.com/robocomp/robocomp/blob/development/doc/README.md)
[Robocomp Contribution guide](https://github.com/robocomp/robocomp/blob/development/CONTRIBUTING.md)

**Good luck with your first issue!**

vkrm1612, 2021-03-11:
I am willing to work on it

vkrm1612, 2021-03-11:
Hi i forked the issue can i know how to open .github file in my local machine 
and what i should replace.....! broken link i couldnt understand....!please help me

arshPratap, 2021-03-11:
Hello Robocomp community, Can I work on this issue?Thanks

vangorade, 2021-03-12:
take

vangorade, 2021-03-12:
anyone working on it? or should I take it. it would be a great first issue for me in this repo

manju838, 2021-03-14:
I am willing to work on this issue.Any help or advice would be appreciated

pawanw17, 2021-03-14:
Hello @vkrm1612 @arshPratap @vangorade and @manju838, this issue has been taken care of, you guys can look into other good-first-issues to get started.

closed 2021-03-14 by pawanw17

HassanRahim26, 2021-03-14:
Hello, @pawanw17, What about my Pull Request? (https://github.com/robocomp/robocomp/pull/320)

PR #320 (2021-03-14): Added a link to a Doc-Tutorial and a YouTube-Tutorial on how to make a PR using Terminal
Fixed #311.
I have added a link to a Doc-Tutorial as well as a Youtube-Tutorial on how to make a PR using Terminal.