robocomp/robocomp #312: [Newcomers issue]

orensbruli, 2021-02-25:
### Describe the bug
 
Clean typos in the docker documentation and check english translation:
https://github.com/robocomp/robocomp/blob/development/docker/Readme.md

### Instructions
This is a ``first-timer-only`` issue. **So the people who have not contributed to robocomp's codebase before, or people who have not contributed to open source before can take this up.**

If you have contributed before, **consider leaving this PR for someone new** :)

#### Prerequisites
You do not need to know anything prior to taking up this issue! Welcome to opensource!

#### 📋 Steps
- [ ] **Claim this issue** : Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ]  **Fork the robocomp/robocomp repository** using the fork button on the topright corner.

- [ ] **Clone the newly forked repository** on your local machine.

- [ ] **Update** the file README.md by removing the broken link.

- [ ] **Commit and push** your changes.

- [ ] **Make a Pull Request** by going to your fork of robocomp.

- [ ] **Ask in comments for a maintainer to review**.

If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow on how GitHub works.](https://guides.github.com/activities/hello-world/)

If you are familiar with the terminal or would like to learn to use it, [here is a great tutorial on how to send a pull request using the terminal.](https://egghead.io/series/how-to-contribute-to-an-open-source-project-on-github)

### Questions?
Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below or ask in the [gitter channel](https://gitter.im/robocomp).

[Robocomp FAQ](https://github.com/robocomp/robocomp/blob/development/doc/FAQ.md)
[Robocomp Tutorials](https://github.com/robocomp/robocomp/blob/development/doc/README.md)
[Robocomp Contribution guide](https://github.com/robocomp/robocomp/blob/development/CONTRIBUTING.md)

**Good luck with your first issue!**

Suhas-G, 2021-02-28:
Hi. I'm Suhas. I'm new to Robocomp. I went through the Issues list, and was wondering if I can take this up?

Also, one of the step says to 'Update the file README.md by removing the broken link', but all the links in [README.md](https://github.com/robocomp/robocomp/blob/development/docker/Readme.md) seem to be proper.

orensbruli, 2021-02-28:
Hello @Suhas-G , welcome to @robocomp
If it's your first time creating a fork and submitting a Pull Request, it's yours. If you already have some experience with it, let this issue for more inexperienced people. Thank you.

> Also, one of the step says to 'Update the file README.md by removing the broken link', but all the links in README.md seem to be proper.

Suhas-G, 2021-02-28:
Hello @orensbruli. I'm not particularly new to Github, even though I haven't contributed to a big open source project before. But maybe I should try a bit more complicated issue :smile:

arshPratap, 2021-03-11:
Hello Robocomp community , I was looking forward to contributing to the organization. Can I work on this issue? Thanks in advance!!

orensbruli, 2021-03-12:
Thank you for the PR. You have some suggestions.

Varun270, 2021-03-13:
@orensbruli  I wanted to know whether this issue is fixed or not , since I am also a newbie in open source . If its open I request you to assign me this issue .

Varun270, 2021-03-13:
@orensbruli  I checked the docker document file and there are still many typos present at the moment , I am willing to solve this issue with your permission.

Fred-JD, 2021-03-14:
Hello, I am new to opensource can I have this new comers issue a try. Thanks

orensbruli, 2021-03-14:
Hello @Fred-JD, thank you for your interest.
There's currently two conflicting Pull Request on this issue. I think It's better to look for other issues. Best!

orensbruli, 2021-03-14:
Tempary clossing issue until pending PR is reviewed.

closed 2021-03-14 by orensbruli