robocomp/robocomp #324: missing dependency in README.md

pawanw17, 2021-03-14:
### Describe the bug
README.md has a missing dependency called argcomplete
this can be overcomed by adding python3-argcomplete in the installation steps

### Your environment
 - Ubuntu/Linux version: [e.g. Ubuntu 18.04]
 - Python / GCC Version: [e.g. Python 3.6]
 

### To Reproduce
Steps to reproduce the behavior:
1. rccomp list

Varun270, 2021-03-15:
@pawanw17  Helllo, Pawan I am a beginner in open source and would love to solve this bug, can you give a detailed explanation about this bug. That would be a great help from your side.

pawanw17, 2021-03-15:
@Varun270 Recently the README.md file was revamped to remove unused packages and simplify the installation steps, accidently this package argcomplete was also removed, you can make a PR by adding the above mentioned package back in the README.md

Varun270, 2021-03-16:
@pawanw17  So all I have to do is simply include python3-argcomplete in installation steps in the readme file?

pawanw17, 2021-03-16:
@Varun270 Yep.

Varun270, 2021-03-26:
@pawanw17  Is this issue still open mate?

pawanw17, 2021-03-28:
Hello @Varun270 A PR was submitted for this issue, you can look onto other recent open issues or maybe review the PR's submitted and suggest changes.

closed 2021-03-28 by pawanw17