robocomp/robocomp #338: [Newcomers issue] Update libs/dsr/README.md

orensbruli, 2021-04-09:
### Describe the bug

There are new methods, for example, size() method in dsr_api for DSR::Graph.
The documentation needs to be reviewed and fixed ([README.md](https://github.com/robocomp/robocomp/blob/development/libs/dsr/README.md))

### Instructions
This is a ``first-timer-only`` issue. **So the people who have not contributed to robocomp's codebase before, or people who have not contributed to open source before can take this up.**

If you have contributed before, **consider leaving this PR for someone new** :)

#### Prerequisites
You do not need to know anything prior to taking up this issue! Welcome to opensource!

#### 📋 Steps
- [ ] **Claim this issue** : Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ]  **Fork the robocomp/robocomp repository** using the fork button on the topright corner.

- [ ] **Clone the newly forked repository** on your local machine.

- [ ] **Commit and push** your changes.

- [ ] **Make a Pull Request** by going to your fork of robocomp.

- [ ] **Ask in comments for a maintainer to review**.

If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow on how GitHub works.](https://guides.github.com/activities/hello-world/)

If you are familiar with the terminal or would like to learn to use it, here is a great tutorial on how to send a pull request using the terminal :-
[Doc-Tutorial](https://github.com/robocomp/robocomp/blob/development/doc/contribute/contribute.md) or  [Video-Tutorial](https://www.youtube.com/watch?v=rgbCcBNZcdQ&t=8s)

### Questions?
Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below or ask in the [gitter channel](https://gitter.im/robocomp).

[Robocomp FAQ](https://github.com/robocomp/robocomp/blob/development/doc/FAQ.md)
[Robocomp Tutorials](https://github.com/robocomp/robocomp/blob/development/doc/README.md)
[Robocomp Contribution guide](https://github.com/robocomp/robocomp/blob/development/CONTRIBUTING.md)

**Good luck with your first issue!**

Varun270, 2021-04-09:
@orensbruli  I would love to solve this issue.

Varun270, 2021-04-09:
@orensbruli  I didn't get it what are the changes do you want me to do like should I have to do a grammar check for README or add some new methods. Please explain I am kind of lost.

orensbruli, 2021-04-09:
Hello @Varun270  and wlecome!
Look https://github.com/robocomp/robocomp/blob/development/libs/dsr/api/dsr_api.cpp#L802
Can you see that there's an undocumented DSRGraph::size() method? I needed it today and looked for it in the README.md of libs/dsr/ and it's not there.
What I suggest is that someone could look into the code and check what other methods are missing in the README.md and write a line or two about its functionality, at least for the obvious ones.
Thank you!

joshboye, 2021-07-11:
Hey @Varun270, just wanted to if you are still working on this issue or else would love to work on this.

Varun270, 2021-07-14:
> Hey @Varun270, just wanted to if you are still working on this issue or else would love to work on this.

Hey @joshboye as of now I am not working on this issue it's all yours!!!

Azhrdev, 2021-11-19:
If anyone is not working on this issue, I would like to work on it

Satvik-Verma, 2022-01-06:
Hey @joshboye, are you still working on this issue?
If not I'd like to work on this.

joshboye, 2022-01-06:
> Hey @joshboye, are you still working on this issue?
> If not I'd like to work on this.

Hey I am not... Please feel free to work on it.

Satvik-Verma, 2022-01-06:
> > Hey @joshboye, are you still working on this issue?
> > If not I'd like to work on this.
> 
> Hey I am not... Please feel free to work on it.

Okay Thank you, I'll work on it!

rinkydevi, 2022-02-05:
@Satvik-Verma hey! are u working on this issue? If not i's like to work on this.

Satvik-Verma, 2022-02-06:
> @Satvik-Verma hey! are u working on this issue? If not i's like to work on this.

Hi @rinkydevi I'm currently caught up with some other work so please feel free to work on this.

rinkydevi, 2022-02-06:
hey @Satvik-Verma ! Thanks for your response,
I will continue for this issue.

rinkydevi, 2022-02-06:
hey @orensbruli !
I am a bit new to github and this is my first issue , first of all thank u so much for providing so much help to the newcomers!
and  i did read the issue and tried to understand a bit of it.
what i understood so far is that i might need to go through the code and update the readme about what all new functions has been added to the code, please tell me if i m going right or i need to understand something and do i need to go through the whole code of the project?
thank u again!

Neel-Shah-29, 2022-02-10:
I would like to fix up this issue , if no ongoing work is going on

mohsinmahmood12, 2022-02-25:
I would like to fix up this issue.

Shobuj-Paul, 2022-03-10:
@mohsinmahmood12 Are you still working on this issue? If not, I would like to take this up.

rakesh-banni, 2022-09-04:
i would like to work on this issue.

deepu718, 2023-01-21:
I am new to open source can I work on this issue if no one is working on this?

VARUN-009, 2023-01-22:
I wanna work on this issue. Please assign this to me and since I'm new to this repo so I request you to please guide me through.

MashiatK, 2023-02-04:
I want to work on the issue as I am new to open source!

suhani-okok, 2023-11-05:
Hey, I want to work on this issue. Please assign me this issue. I'm new to open source contribution please do guide me. thankyou.

AADI-234, 2023-11-28:
@rinkydevi are you working on the issue or not
if  not , i would love to work on it