robocomp/robocomp #341: [BUG] Components not able to implicitly locate RoboComp specific shared libraries

nocoinman, 2021-04-23:
### Describe the bug
Building certain components from source resulting in failure because of linking errors. Specifically cmake isn't able to locate some robocomp specific internal libraries and explicit declaration of these libraries in cmake seems to be required.

### Your environment
 - Ubuntu/Linux version: Ubuntu 20.04
 - Python Version: Python 3.8.5
 - GCC Version: gcc version 9.3.0
 - RoboComp built from source
        - Branch - [development](https://github.com/robocomp/robocomp/tree/development)
        - Commit - https://github.com/robocomp/robocomp/commit/cdc448fdd47a07cf0d75948b4f3b3ef21cb72057
     
 - [ ] Robocomp is compiled with FCL support.
 - [x] DSR dependencies have been installed.  

### To Reproduce

Issue was first encountered while trying to build [robocomp/optimizer](https://github.com/robocomp/optimizer) components, specifically the [comp-one](https://github.com/robocomp/optimizer/tree/main/comp-one) component. However the current version of this repo has some other bugs so I recommend you use my fork instead to reproduce the bug of interest [nocoinman/optimizer](https://github.com/nocoinman/optimizer/tree/internal-lib-linking-issue).

Steps to reproduce the behavior:
1. Clone the `internal-lib-linking-issue` branch of [nocoinman/optimizer](https://github.com/nocoinman/optimizer).
2. Build the *comp-one* component

```bash 
rcbuild comp-one
```

NOTE: *The said component depends on some third party libraries like [cppitertools](https://github.com/ryanhaining/cppitertools) and [Gurobi](https://www.gurobi.com/downloads/gurobi-optimizer-eula/).*

### Expected behavior
CMake throws linking errors

```bash
/usr/bin/ld: CMakeFiles/compone.dir/specificworker.cpp.o: in function `SpecificWorker::compute_path()':
specificworker.cpp:(.text+0x219e): undefined reference to `RMat::QVec::vec3(float, float, float)'
/usr/bin/ld: specificworker.cpp:(.text+0x21d1): undefined reference to `InnerModel::transform(QString const&, RMat::QVec const&, QString const&)'
/usr/bin/ld: CMakeFiles/compone.dir/specificworker.cpp.o: in function `SpecificWorker::local_controller(std::vector<QPointF, std::allocator<QPointF> > const&, std::vector<RoboCompLaser::TData, std::allocator<RoboCompLaser::TData> > const&, QPointF const&, QPointF const&)':
specificworker.cpp:(.text+0x2562): undefined reference to `RMat::QVec::vec3(float, float, float)'
/usr/bin/ld: specificworker.cpp:(.text+0x2592): undefined reference to `InnerModel::transform(QString const&, RMat::QVec const&, QString const&)'
/usr/bin/ld: CMakeFiles/compone.dir/specificworker.cpp.o: in function `SpecificWorker::read_base()':
specificworker.cpp:(.text+0x28e9): undefined reference to `InnerModel::updateTransformValues(QString, float, float, float, float, float, float, QString)'
/usr/bin/ld: CMakeFiles/compone.dir/specificworker.cpp.o: in function `SpecificWorker::setParams(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, RoboCompCommonBehavior::Parameter, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, RoboCompCommonBehavior::Parameter> > >)':
specificworker.cpp:(.text+0x63e4): undefined reference to `InnerModel::InnerModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)'
/usr/bin/ld: CMakeFiles/compone.dir/specificworker.cpp.o: in function `SpecificWorker::compute()':
specificworker.cpp:(.text+0xa536): undefined reference to `RMat::QVec::vec3(float, float, float)'
/usr/bin/ld: specificworker.cpp:(.text+0xa581): undefined reference to `InnerModel::transform(QString const&, RMat::QVec const&, QString const&)'
/usr/bin/ld: specificworker.cpp:(.text+0xa65c): undefined reference to `RMat::QVec::dotProduct(RMat::QVec const&) const'
/usr/bin/ld: CMakeFiles/compone.dir/specificworker.cpp.o: in function `std::_Sp_counted_ptr_inplace<InnerModel, std::allocator<InnerModel>, (__gnu_cxx::_Lock_policy)2>::_M_dispose()':
specificworker.cpp:(.text._ZNSt23_Sp_counted_ptr_inplaceI10InnerModelSaIS0_ELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv[_ZNSt23_Sp_counted_ptr_inplaceI10InnerModelSaIS0_ELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv]+0x9): undefined reference to `InnerModel::~InnerModel()'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/compone.dir/build.make:388: ../bin/compone] Error 1
make[1]: *** [CMakeFiles/Makefile2:144: src/CMakeFiles/compone.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

### Additional context

**The said error seems to be caused because the build system isn't implicitly linking internal libraries like libqmat.so and libinnermodel.so.** Explicitly declaring these in the `CMakeListsSpecific.txt` of the component resolves the issue as shown [here](https://github.com/nocoinman/optimizer/blob/3a89939bd6bafdf45f31d9e6384f50a456a050da/comp-one/src/CMakeListsSpecific.txt#L35). In my system these libraries are installed at `/usr/local/lib`

### If the bug is confirmed, would you be willing to submit a PR?
No

nocoinman, 2021-04-26:
I just want to point out that this is starting to be a concern for building components that are mentioned in the tutorials. It is really hard to follow them now as the mentioned components aren't able to find many of robocomps internal libraries.

orensbruli, 2021-04-26:
I have asked other members of Robocomp their opinion about if all the libraries should be included in the compilation of a component by default and generated in the CMakeList files with robocomodsl. 
While it's decided, can you point to all the components that you have found failing?

nocoinman, 2021-04-26:
It's not that all libraries should be compiled. When robocomp is built these libraries are built by default. An example would be `libqmat.so` and `libinnermodel.so`. Similarly I found issues with `libdsr_gui.so` and other libraries in DSR. The problem is these library destinations are not included by default during compilation of components and we need to explicitly state in cmakelists to link these libraries.