robocomp/robocomp #385: [BUG] Invalid URL in Tutorials

shivangvijay, 2023-01-08:
### Describe the bug
Link for "Starting with DSR" is not valid.

### Your environment
 - Ubuntu/Linux version: Ubuntu 20.04
 - GCC Version: 9.4.0

### To Reproduce
Steps to reproduce the behavior:
1. Go to https://github.com/robocomp/robocomp/blob/development/doc/README.md
2. Click on "Starting with DSR" section

### Screenshots

[image]

### If the bug is confirmed, would you be willing to submit a PR?
Yes

Rohan-Dah, 2023-01-28:
Hey.. I would like to work on this issue... What will be the correct redirecting URL, can you please provide that link?

kannishk, 2023-02-20:
I would like to work on this issue, assign me!

Andrew-Doghry, 2023-02-25:
Hi I would like to work on this issue can you assign me ?